CohenLabPrinceton/pvp #10: Adding get_controls function

sunnyszy, 2020-04-21:
Besides the `set_controls()` interface, we have in coordinator and controller, I feel we should also define `get_controls()` interface. This allows:

- Easier unit test for `set_controls()`
- After set, the coordinator/controller can immediately call get to verify the set results. This gives a consistency guarantee. 

My current definition: 
```
def get_controls(self, control_setting_name: ControlSettingName) -> ControlSettings:
```

sunnyszy, 2020-04-21:
If you guys feel okay, @mschottdorf can you add the function to the controller?

gdoubleyew, 2020-04-21:
Sounds good to me

mschottdorf, 2020-04-21:
yep, sure!

sunnyszy, 2020-04-21:
@mschottdorf FYI I added that yesterday as it is only copying your get_controls implementation :D

closed 2020-04-21 by sunnyszy