CohenLabPrinceton/pvp #27: Value unification

sneakers-the-rat, 2020-04-29:
Going to use this issue to keep track of the places where attribute/variable names need to take from `values`

`common.message.SensorValues` - https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/35ee8be49741ad9a060cf76c650231686f8c8b37/vent/common/message.py#L5

`common.message.ControlSettingName` and many of its uses:
* coordinator: https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/860f66821414aaf037c273a83be0f9bfdefd707b/vent/coordinator/coordinator.py#L113
* control module get: https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/27885ac510300178d9b47b970dc492dbbb831fa8/vent/controller/control_module.py#L292
* control module set: https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/27885ac510300178d9b47b970dc492dbbb831fa8/vent/controller/control_module.py#L327

have a commit with some examples of how we might do it:
https://github.com/CohenLabPrinceton/Ventilator-Dev/commit/a47443b779e184289f57c95d31e99442b4a29a87

sunnyszy, 2020-04-29:
Would https://github.com/CohenLabPrinceton/Ventilator-Dev/pull/26 look good to you?

sneakers-the-rat, 2020-04-30:
left some comments in that PR, i think this issue will be pretty stale by the time we get around to addressing these. don't wanna b pushy ;)

closed 2020-04-30 by sneakers-the-rat