CohenLabPrinceton/pvp #32: Timeouts on get_ and set_ functions with decorators?

mschottdorf, 2020-05-01:
How do we set timeouts universally? Should we use decorators? Could use something along these lines:
```
import signal
import time
from contextlib import contextmanager

class TimeoutException(Exception): pass

@contextmanager
def time_limit(seconds):
    def signal_handler(signum, frame):
        raise TimeoutException("Timed out!")
    signal.signal(signal.SIGALRM, signal_handler)
    signal.alarm(seconds)
    try:
        yield
    finally:
        signal.alarm(0)

def timelimitdecorator(func):
    def wrapper(*args, **kwargs):
        try:
            with time_limit(10):
                func(*args, **kwargs)
        except TimeoutException as e:
            print("Timed out!")
    return wrapper
```

Then on a function
```
@timelimitdecorator
def foo(sleeptime):
    time.sleep(sleeptime)
    print("hello")
```

We get a timeout on foo(11), but not on foo(1).

sunnyszy, 2020-05-01:
I would propose that let's think about what functions need a timeout. I feel GUI <-> coordinator, coordinator <-> controller don't really need it as they are within the same process, and the function is basically returned a copy of the value.

Coordinator IPC definitely needs a timeout. And does controller <-> sensor need a timeout?

mschottdorf, 2020-05-03:
I guess, we'll have to see how the sensor-abstraction layer turns out

sunnyszy, 2020-05-04:
Yes, I agree. I would suggest let's hold this issue a bit until we have more information about the requirement and the assumption about the sensor layers. To give a more concrete example, if the sensor call blocking probability is ~1 failure/month, it might not worth implementing this complexity into the logic. Remind that the coordinator will restart the controller process if it hangs.

sneakers-the-rat, 2020-05-04:
somehow didn't see this, v clever. think this is a great idea (although i would name the decorator `timeout` :) )

PR #46 (2020-05-08): Draft implementation of @properties for locks & exceptions
This is a rough draft - mostly untested and I didn't go through the project to apply @locked. exception logger doesn't actually log anything, just raises whatever it's passed.

Nevertheless, I figured I'd open up a feature branch & draft PR to collect thoughts & code. 

closes #32 
closes #35 
closes #38
closed 2020-05-08 by mschottdorf