CohenLabPrinceton/pvp #34: IndexError: empty axes in `control_module.__analyze_last_waveform`

sneakers-the-rat, 2020-05-02:
Have had this happen a few times if i leave the program running for a few minutes in the background. branch is current w/ https://github.com/CohenLabPrinceton/Ventilator-Dev/commit/32d64d05f622e445633393847e63ef8afbfd11f9

I think we should have the tests run for longer, ive seen a few errors come up only after the program has been running for a few minutes. or is there some way to make control module simulation generate values as fast as possible and not simulate a sampling rate?

```
Traceback (most recent call last):
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/jonny/git/Ventilator-Dev/vent/controller/control_module.py", line 714, in _start_mainloop
    self._PID_update(dt = dt)                               # Update the PID Controller
  File "/Users/jonny/git/Ventilator-Dev/vent/controller/control_module.py", line 500, in _PID_update
    self.__analyze_last_waveform()    # Analyze last waveform
  File "/Users/jonny/git/Ventilator-Dev/vent/controller/control_module.py", line 251, in __analyze_last_waveform
    self._DATA_PEEP = np.percentile(pressure[ pressure < np.mean(pressure)], 20 )
  File "<__array_function__ internals>", line 6, in percentile
  File "/usr/local/lib/python3.7/site-packages/numpy/lib/function_base.py", line 3713, in percentile
    a, q, axis, out, overwrite_input, interpolation, keepdims)
  File "/usr/local/lib/python3.7/site-packages/numpy/lib/function_base.py", line 3833, in _quantile_unchecked
    interpolation=interpolation)
  File "/usr/local/lib/python3.7/site-packages/numpy/lib/function_base.py", line 3410, in _ureduce
    r = func(a, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/numpy/lib/function_base.py", line 3948, in _quantile_ureduce_func
    x1 = take(ap, indices_below, axis=axis) * weights_below
  File "<__array_function__ internals>", line 6, in take
  File "/usr/local/lib/python3.7/site-packages/numpy/core/fromnumeric.py", line 194, in take
    return _wrapfunc(a, 'take', indices, axis=axis, out=out, mode=mode)
  File "/usr/local/lib/python3.7/site-packages/numpy/core/fromnumeric.py", line 61, in _wrapfunc
    return bound(*args, **kwds)
IndexError: cannot do a non-empty take from an empty axes.```

mschottdorf, 2020-05-02:
Thanks for finding! I'll try to reproduce it!

mschottdorf, 2020-05-02:
I'm not sure what caused it, but I had a few dead stubs left over when I transitioned from the ring-buffer to deque. I've removed all this in the new PR. Might have resolved this bug as well.

I ran the test 20x, and let the code run for ~330.000 cycles. This all worked well.

[image]

sneakers-the-rat, 2020-05-03:
sorry for not providing reproducible error, thanks for checking it out. i'll let ya know if anything else pops up but by and large the simulation has worked well :)

sneakers-the-rat, 2020-05-05:
same thing happened again w/ but only after like 15m https://github.com/CohenLabPrinceton/Ventilator-Dev/commit/17504a8914445993811f5abcf2b9fbd4a2d92769 

going to run again w/ pdb for this and the complex number error and get back 2 u

sneakers-the-rat, 2020-05-05:
the gui stays alive, but the controller dies. seems like controller should be able to swallow numerical errors -- exception handling in the main loop

mschottdorf, 2020-05-05:
Yeah, can be done (just return a np.nan for peep and pip), but I really want to know want is going on here... might be something important

sneakers-the-rat, 2020-05-05:
traced it back to the `Qin` value from `SimulatedPropValve`

sneakers-the-rat, 2020-05-05:
actually wait no `x` in the line before is nan first

mschottdorf, 2020-05-05:
I'm pretty sure that can only happen if the pressure is np.nan - sounds like a bug in the balloon simulation. Can you give me parameters to try to reproduce this here?

sneakers-the-rat, 2020-05-05:
yes i believe the balloon is having some beef with natural law:

```python
pprint(self.Balloon.__dict__)
{'P0': 0,
 'PC': 20,
 'current_flow': -0.9999999999999993,
 'current_pressure': nan,
 'current_volume': -2.6030784139434933,
 'delay': False,
 'fio2': 13990.878703428589,
 'humidity': 77.49196868074476,
 'leak': True,
 'max_volume': 6,
 'min_volume': 1.5,
 'r_real': nan,
 'temperature': 362.6075413405974}```

sneakers-the-rat, 2020-05-05:
`np.clip` volume to min zero?

sneakers-the-rat, 2020-05-05:
clamped volume to min 0 and i'm getting some cool other stuff like temperature shooting off to a billion. i think it might be that the simulation expects `dt` to be very small and having a large `dt` causes big error. i get the bug most when I minimize and remaximize the GUI window so it could be that the simulation hangs while that long draw is happening. all the more important to get into separate processes asap

sneakers-the-rat, 2020-05-05:
dk if that's just a simulator problem or if that's in the control logic, if it's just simulator i don't know that we need to spend a ton of time making it bulletproof. I do think that having erratic `dt`s would be a good test though.

mschottdorf, 2020-05-05:
I'll spend some time today on the simulator. This should be fixed... I think you're right, and this is likely a problem in the calculation of the OU process with large dt. There are a few other things in the simulator, that need fixing as well. E.g. controlled by solenoids, flows should be proportional to pressure differences. I'll add this right now

sneakers-the-rat, 2020-05-05:
ok cool. 10m runtime and the problem hasn't returned

sneakers-the-rat, 2020-05-05:
but ya in `Balloon.update` I just went like

```python
    def update(self, dt):  # Performs an update of duration dt [seconds]
        self.current_volume += self.current_flow * dt
        self.current_volume  = np.max([0, self.current_volume])
```

mschottdorf, 2020-05-05:
[image]
look at that leaky balloon... that cannot be right!

mschottdorf, 2020-05-05:
Alright, I've made a PR with improved balloon physics. Hopefully, clipping the volume at zero is not necessary anymore - it shouldn't happen

sneakers-the-rat, 2020-05-05:
idk much about balloons but that balloon has got some attitude

mschottdorf, 2020-05-07:
PR is merged

closed 2020-05-07 by mschottdorf