CohenLabPrinceton/pvp #35: Locks handled as @properties and a careful programming proposition

sneakers-the-rat, 2020-05-02:
**Note: this is something I think we should save effort on until we get a working prototype out, but want to write it down**

Reading through the code in a bit more detail while setting up alarms, and some of this locking is making me nervous. Please understand my intentions as genuine and goode and not trying to code shame. I raised this during our last call and just wanted to follow up

**Problem:**
locks, well, lock. and so if a lock is ever acquired without being released, and there are no timeouts, then the program hangs. The following code makes me nervous:

https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/32d64d05f622e445633393847e63ef8afbfd11f9/vent/coordinator/coordinator.py#L142
```python
while not self.stopping.is_set():
    sensor_values = self.control_module.get_sensors()
    self.lock.acquire()
    self.COPY_sensor_values = sensor_values
    self.last_message_timestamp = sensor_values.timestamp
    self.lock.release()
    for name in [ValueName.PIP,
                 ValueName.PIP_TIME,
                 ValueName.PEEP,
                 ValueName.BREATHS_PER_MINUTE,
                 ValueName.INSPIRATION_TIME_SEC]:
        self.lock.acquire()
        in_control_settings = name not in self.COPY_control_settings
        self.lock.release()
        if in_control_settings:
            control_setting = self.control_module.get_control(name)
            self.lock.acquire()
            self.COPY_control_settings[name] = control_setting
            self.lock.release()

        self.lock.acquire()
        disagreed_tentative = name in self.COPY_tentative_control_settings and self.COPY_tentative_control_settings[name] != self.COPY_control_settings[
                    name]
        self.lock.release()
        if disagreed_tentative:
            self.lock.acquire()
            tentative_control_setting = self.COPY_tentative_control_settings[name]
            self.lock.release()
            self.control_module.set_control(tentative_control_setting)
            self.lock.acquire()
            self.COPY_control_settings[name] = self.control_module.get_control(name)
            self.lock.release()
    # sleep 10 ms
    time.sleep(0.01)
```

Given the number of times the lock is accessed, and the degree of nesting, this seems like a pretty difficult to maintain/error-prone method of thread management. 

**Proposal:**
From what I understand, the lock is just acquired and released whenever any of these set of attributes are accessed. 

let's use properties, this is what they were born to do.

Observe the following extraordinarily careful pattern that I believe we should follow as a general pattern for repetitively called methods

```python
import traceback
import functools

# some global stack param
MAX_STACK_DEPTH = 20

def locked(func):
    """
    Wrapper to use as decorator, handle lock logic for a
    @property

    Arguments:
        func (callable): function to wrap
    """

    # define a function to return that wraps the inner
    # function around some standardized error handling
    # logic

    # use .wraps to preserve inner method identity
    @functools.wraps(func)
    def lock_wrapper(self, *args, **kwargs):
        ret = None

        try:
            self.lock.acquire()
            ret = func(self, *args, **kwargs)
        except Exception as e:
            log_exception(
                e, 
                traceback.extract_tb(MAX_STACK_DEPTH)
            )
        finally:
            self.lock.release()

        return ret

    return lock_wrapper

class CoordinatorBase:
    def __init__(self, sim_mode=False):

        self.lock = threading.Lock()

        self.control_module = get_control_module(sim_mode)
        self._sensor_values = None

    @locked
    @property
    def sensor_values(self):
        return self._sensor_values

    @locked
    @sensor_values.setter
    def sensor_values(self, sensor_values):
        self._sensor_values = sensor_values
```

This would allow every lock acquisition and release to be replaced with standard attribute set/get methods. ie instead of
`
# set
self.lock.acquire()
self.COPY_sensor_values = ...
self.lock.release()

# get
def get_sensors(self) -> SensorValues:
    self.lock.acquire()
    sensor_values = self.COPY_sensor_values
    self.lock.release()
    # return res
    return sensor_values
`
(which is also retrieved by an additonal method `get_sensors` that acquires/releases the lock) 

you just use
`
# set
self.sensor_values = ...

# get
self.sensor_values
`

The benefits are of course not just aesthetic, but guarantee that the lock is always released due to the use of `finally` and that any exceptions are always logged.

sneakers-the-rat, 2020-05-02:
sorry laptop battery dying and wanted to post to save before i could finish commenting code!!!

mschottdorf, 2020-05-03:
makes perfect sense to me, had suggested something similar for timeouts. Let's talk about this in our next meeting, but I'd love to adapt this soon. Will make the code much cleaner!

PR #46 (2020-05-08): Draft implementation of @properties for locks & exceptions
This is a rough draft - mostly untested and I didn't go through the project to apply @locked. exception logger doesn't actually log anything, just raises whatever it's passed.

Nevertheless, I figured I'd open up a feature branch & draft PR to collect thoughts & code. 

closes #32 
closes #35 
closes #38
closed 2020-05-08 by mschottdorf