CohenLabPrinceton/pvp #38: Error handling in vent.io

CommReteris, 2020-05-03:
Need to make sure all possible exceptions & hangs are caught.

See: [This](https://github.com/CohenLabPrinceton/Ventilator-Dev/pull/31#discussion_r418366771) and 
[this.](https://github.com/CohenLabPrinceton/Ventilator-Dev/pull/31#discussion_r418367429)

sneakers-the-rat, 2020-05-03:
proposed a generalized method of doing this here:
https://github.com/CohenLabPrinceton/Ventilator-Dev/issues/35

give it a shot, haven't had a minute to actually test it, just a draft. All it requires is for us to build some logging system. The logging module is solid and I can draft up a way of spawning logging objects for each of the separate modules.

sneakers-the-rat, 2020-05-03:
scuse me - that general pattern of having a decorator to wrap sensitive calls in a try/except/finally block, sensor readings would probs look a bit different since they don't need locking.

PR #46 (2020-05-08): Draft implementation of @properties for locks & exceptions
This is a rough draft - mostly untested and I didn't go through the project to apply @locked. exception logger doesn't actually log anything, just raises whatever it's passed.

Nevertheless, I figured I'd open up a feature branch & draft PR to collect thoughts & code. 

closes #32 
closes #35 
closes #38
closed 2020-05-08 by mschottdorf