CohenLabPrinceton/pvp #40: Complex numbers

sneakers-the-rat, 2020-05-04:
The simulator sometimes breaks and starts spitting out complex values for pressure:

```
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/jonny/git/Ventilator-Dev/vent/controller/control_module.py", line 716, in _start_mainloop
    self._PID_update(dt = dt)                               # Update the PID Controller
  File "/Users/jonny/git/Ventilator-Dev/vent/controller/control_module.py", line 469, in _PID_update
    if self._DATA_PRESSURE < 1.1*self.__SET_PEEP:
TypeError: '<' not supported between instances of 'complex' and 'float'
```

tracing it back the complex value gets assigned here:
https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/87a83f5964ffbc1f40c558b07716e6f3977ca44c/vent/controller/control_module.py#L620

but i have no idea why it would be float sometimes and not other times....

mschottdorf, 2020-05-04:
intriguing.. Could be related to negative volumes, i.e. small errors if more air flows out than in. Will investigate. Thanks!

mschottdorf, 2020-05-05:
This might have been related to the other issue with tormented physics

mschottdorf, 2020-05-07:
PR is merged, fingers crossed!

closed 2020-05-07 by mschottdorf