CohenLabPrinceton/pvp #41: Implementing Alarms

sneakers-the-rat, 2020-05-04:
Making a single place to talk about implementing alarms (as it applies to code, the slack channel is more for organizing w/ rest of team). I'll be editing this post as I work so it may change

My current understanding of what we had planned
* Controller and GUI both check for alarms, GUI serves as a double-check on alarm checking from controller (did we both get an alarm just now?) and otherwise takes its alarms from controller. 
* Alarms will be 'sticky' and be displayed until user dismisses them
* User will have option to mute alarms for specific time - i'm thinking all alarms, and alarms for a particular value.
* ... more but can't remember good

Questions
* logging? who logs the alarms. I think it sorta makes sense to have the GUI log them since the GUI will be the one displaying, generating text for them, and dismissing them, but also see value in dual-logging (`logging` module makes it easy to have multiple logs for different objects)
* sound? I can make some alarm sounds (in a former life i was a sound engineer for ~7 years & apprenticed with a sound designer) but figure if we can use alarm sounds that would be familiar it would be better

sneakers-the-rat, 2020-05-05:
Currently the user sets the desired value of PIP as a control, but the value returned by `get_sensors` is pressure, which is also the value that users set limits on in the gui. Do we
* map the safe_range of pressure onto PIP, PEEP, etc? (eg. max range of pressure is max allowed pip?
or
* have `get_sensors` return measured PIP, PEEP, etc. and give each of them a `monitor` object to set ranges?

mschottdorf, 2020-05-05:
For the protocol, the controller is logging past alarms (and past waveforms) up to 100 entries. It would be good to also save this sporadically in a file, possibly by the coordinator

sneakers-the-rat, 2020-05-08:
I was looking for how to send down limits on monitored values like `FiO2` and as far as I can tell there isn't such a thing in the controller. It looks like a lot of the variables that manage alarms in the controller are hardcoded. This makes it so it can't trigger an alarm for, eg. FiO2 going out of range. It also means that the controller needs to have its code modified every time there is a new value to check -- eg. if we want to trigger an alarm off of `IE_RATIO` . The gui has all of its alarm ranges generated programmatically from `values.py`, so this is not the case. I see two options, a) have the controller and gui both generate alarms and have the coordinator house an `alarm_manager` that can handle the logic of combining them, b) the controller can be rewritten such that it programmatically updates variables from `values.py` s.t. it can generate alarms for all of the alarmed values.

these are the places i found with hardcoded variables:

https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/c7fddba3528d2c049805b97804308019f28d255f/vent/controller/control_module.py#L100
https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/c7fddba3528d2c049805b97804308019f28d255f/vent/controller/control_module.py#L146
https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/c7fddba3528d2c049805b97804308019f28d255f/vent/controller/control_module.py#L303

this would be pretty trivial to implement. it looks like you need, for each, value, min, max, and lastset. in your `__init__` it could look like

```python
self.controls = {}
for val_name, val_params in values.CONTROL.items():
    self.controls[val_name] = {
        'value': None,
        'min': val_params['safe_range'][0],
        'max': val_params['safe_range'][1],
        'lastset': time.time()
```

you could make the different versions you have for each of the variables, eg `COPY` and the like as well.

so then the rest of your methods could be updated to follow that pattern. like `set_control`:

```python
if control_setting.name in ValueName:
    self.controls[control_setting.name]['value'] = control_setting.value
    self.controls[control_setting.name]['min'] = control_setting.min
```

and likewise the `__update_alarms` method could be something like 

```python
for control_name, control_params in self.controls.items():
    self.__test_critical_levels(min=control_params['min'], max = control_params['max'], value=control_params['value'], name = control_name)
```

mschottdorf, 2020-05-08:
makes perfect sense to me. Yes, the code does not allow to send down alarms (yet)

sneakers-the-rat, 2020-05-12:
I'm thinking that basically all alarm checking logic should come out of the controller except for the high-pressure response. I'll make the gui and controller reciprocal so that the gui also pulls controls from the controller so that if the controller drops PIP the gui will reflect that. 

as i mentioned in the meeting earlier today, there's another way of categorizing values, `continuous` or `phasic` (idk that terminology doesn't seem right) - measurements that are made continuously like pressure and those that are computed once per breath cycle. I'm going to add that as an additional parameterization of a value, and also keep track of the breath cycle number and send that in `SensorValues`. that'll be necessary for defining alarm conditions based on breath cycle values

mschottdorf, 2020-05-12:
The controller is also getting quite complex. Would be great to outsource some of that alarm code

sneakers-the-rat, 2020-05-14:
Currently i have two alarm parameters:

if an alarm `latch` es, it a) does not decrement its severity even when the values fall back into a lower-severity range and b) is only cleared when both 1) the user has requested that the alarm be cleared and 2) the alarm values return to the ‘off’ range.
if an alarm is `persistent`, it a) is decremented in severity when values fall into a lower-severity range, b) can be cleared from the gui at any time, but c) will not be removed from the GUI until the user manually clears it (otherwise alarms are cleared when they return to OFF)
is there a case where we wouldn’t want a `persistent` alarm to not `latch` ? otherwise I will remove the `persistent` attribute

sneakers-the-rat, 2020-05-14:
4:51
Currently it is possible to clear an alarm
indefinitely - silencing an alarm will cause that alarm to not be raised again until it returns to OFF severity.
for a period of time - the gui will provide the user a choice of snooze times, and the alarm will not be raised again until the time is up.
they are mutually exclusive — if a timeout is provided, the alarm will not be re-raised if it goes OFF and back on.
should all alarm dismissals be timed? ie. if an alarm is cleared without an explicit snooze, it will not be raised until it goes OFF and back ON or some default time passes?

sneakers-the-rat, 2020-05-19:
preserving some todo items from the pr

https://github.com/CohenLabPrinceton/Ventilator-Dev/pull/64#discussion_r426008278

https://github.com/CohenLabPrinceton/Ventilator-Dev/pull/64#discussion_r426010901

https://github.com/CohenLabPrinceton/Ventilator-Dev/pull/64#discussion_r426011154

https://github.com/CohenLabPrinceton/Ventilator-Dev/pull/64#discussion_r426012813

https://github.com/CohenLabPrinceton/Ventilator-Dev/pull/64#discussion_r426013542

https://github.com/CohenLabPrinceton/Ventilator-Dev/pull/64#discussion_r426016985

https://github.com/CohenLabPrinceton/Ventilator-Dev/pull/64#discussion_r426018333

https://github.com/CohenLabPrinceton/Ventilator-Dev/pull/64#discussion_r426023529

closed 2020-05-29 by sneakers-the-rat