CohenLabPrinceton/pvp #54: Logging

sneakers-the-rat, 2020-05-08:
Let's keep track of what we need to do for logging!!!

considerations
* make a standardized logging class that each object can instantiate, but don't necessarily make a centralized logging object that needs to be fed information.
* size - don't fill up the disk with logs. rotate logs, check how much space is left etc.

mschottdorf, 2020-05-09:
To merge this with the other issue-> Considering the finite live time of the components, and future maintenance of the device, we have to **keep track of open/close cycles**, and **time since installing a particular sensor/valve**.

sunnyszy, 2020-05-15:
Python has a powerful logging system: https://docs.python.org/3.8/howto/logging-cookbook.html
I wonder whether we can leverage that.

Ideally, anything that can be serialized can be logged. But we may not want to log things that need to be read back again (e.g., persistent config).

sunnyszy, 2020-05-19:
I forgot the schedule for this. Will you implement this @mschottdorf? Or I can take a look and implement this.

mschottdorf, 2020-05-19:
I've you have a free afternoon, go for it! I haven't had the time to do it

mschottdorf, 2020-05-19:
My plan was implementing two things:

- python's logging for general logs across threads, mostly because it is thread safe, and 
- A hdf5-system, similar to [jonny's work](https://github.com/wehr-lab/autopilot/blob/2to3/autopilot/core/subject.py) within the controller to keep track of the waveforms.

sunnyszy, 2020-05-19:
Why not I do part one and you do part two? I used logging before.

mschottdorf, 2020-05-19:
go for it; for part 2, I'd probably write a class, and let the instance live in the controller thread. Any objections?

sunnyszy, 2020-05-19:
I second that. It would be easier to make it stateless, otherwise, you have to persist this class as well.

sneakers-the-rat, 2020-05-29:
Closing b/c logging facility has been written.

closed 2020-05-29 by sneakers-the-rat