CohenLabPrinceton/pvp #55: IndexError: cannot do a non-empty take from an empty axes.

mschottdorf, 2020-05-09:
Might be related to Jonny's reported bugs. When re-scaling, or switching between desktops, or running for a long time, the simulator crashes, or simulated temperatures/pressures go haywire.

I'm trying to figures out what is going on.

```
ms81@pni-18c02xj50tjgh5 Ventilator-Dev % ipython ./vent/main.py
qt.qpa.fonts: Populating font family aliases took 248 ms. Replace uses of missing font family "Fira Code" with one that exists to avoid this cost.
/Users/ms81/Ventilator-Dev/vent/controller/control_module.py:706: RuntimeWarning: invalid value encountered in double_scalars
  self.r_real = (3 * self.current_volume / (4 * np.pi)) ** (1 / 3)
/usr/local/lib/python3.7/site-packages/pyqtgraph/graphicsItems/PlotCurveItem.py:153: RuntimeWarning: All-NaN slice encountered
  b = (np.nanmin(d), np.nanmax(d))
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/ms81/Ventilator-Dev/vent/controller/control_module.py", line 805, in _start_mainloop
    self._PID_update(dt = dt)                               # Update the PID Controller
  File "/Users/ms81/Ventilator-Dev/vent/controller/control_module.py", line 522, in _PID_update
    self.__analyze_last_waveform()    # Analyze last waveform
  File "/Users/ms81/Ventilator-Dev/vent/controller/control_module.py", line 264, in __analyze_last_waveform
    self._DATA_PEEP = np.percentile(pressure[ pressure < np.mean(pressure)], 20 )
  File "<__array_function__ internals>", line 6, in percentile
  File "/usr/local/lib/python3.7/site-packages/numpy/lib/function_base.py", line 3706, in percentile
    a, q, axis, out, overwrite_input, interpolation, keepdims)
  File "/usr/local/lib/python3.7/site-packages/numpy/lib/function_base.py", line 3826, in _quantile_unchecked
    interpolation=interpolation)
  File "/usr/local/lib/python3.7/site-packages/numpy/lib/function_base.py", line 3403, in _ureduce
    r = func(a, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/numpy/lib/function_base.py", line 3941, in _quantile_ureduce_func
    x1 = take(ap, indices_below, axis=axis) * weights_below
  File "<__array_function__ internals>", line 6, in take
  File "/usr/local/lib/python3.7/site-packages/numpy/core/fromnumeric.py", line 194, in take
    return _wrapfunc(a, 'take', indices, axis=axis, out=out, mode=mode)
  File "/usr/local/lib/python3.7/site-packages/numpy/core/fromnumeric.py", line 61, in _wrapfunc
    return bound(*args, **kwds)
IndexError: cannot do a non-empty take from an empty axes.
```

mschottdorf, 2020-05-10:
This bug is caused extremely long update-times. Here is a snapshot of the update times (in seconds) per controller iteration just before the controller went crazy:
[image]

I don't know how to fix this (yet), but dt is considerably longer than an entire breath cycle.

sunnyszy, 2020-05-11:
When I replicate this bug, I found another bug that the GUI in my environment won't start. See: https://github.com/CohenLabPrinceton/Ventilator-Dev/issues/60

mschottdorf, 2020-05-13:
With the current PR, the controller can now dead with long waits between updates:

[image]

But we have to figure out why this is happening.

sunnyszy, 2020-05-14:
This looks pretty cool!

sneakers-the-rat, 2020-05-14:
it looks like it can't find the font I added to the repo, so I'll have to troubleshoot that. the reason I packaged a font is because to specify q generic font you have to build the entire font db which is costly, like 5 seconds costly.

mschottdorf, 2020-05-17:
On Grant's suggestion: For the plots above, I was running `coordinator = get_coordinator(single_process=True, sim_mode=True)`. I've done the same `coordinator = get_coordinator(single_process=False, sim_mode=True)`. This seems to solve the timeout-issue. 

In its own thread, and with a goal sleep-time of 10ms between cycles in the controller, the average controller-loop-time was ~11.5ms. This plot is the loop-time density over ~30min of recording (while I was working on my laptop, UI in the background :-) ).

[image]

mschottdorf, 2020-05-21:
I'll close this. Problem was apparently solved by threading

closed 2020-05-21 by mschottdorf