CohenLabPrinceton/pvp #60: UI fails in my Ubuntu desktop

sunnyszy, 2020-05-11:
When running main.py, the UI doesn't start.

Code hash: ce66e4fbfa1f1318c02233c68c52124d893ce2dc
OS: ubuntu 18.04

```
/home/zhenyus/anaconda3/envs/Ventilator-Dev/bin/python /home/zhenyus/Ventilator-Dev/vent/main.py --simulation --single_process
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, minimal, minimalegl, offscreen, vnc, xcb.

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)

```

sneakers-the-rat, 2020-05-14:
definitely an environment issue. try this

https://forum.qt.io/topic/93247/qt-qpa-plugin-could-not-load-the-qt-platform-plugin-xcb-in-even-though-it-was-found/9

sunnyszy, 2020-05-14:
I tried this, however, the answer seems to assume the distribution is archlinux. Let's not worry about this right now, as I don't feel like this is a very important issue.

closed 2020-05-14 by sunnyszy