CohenLabPrinceton/pvp #73: v1.0 - Implementing logging

sneakers-the-rat, 2020-05-29:
We now have a logging module (actually two!) and now we need to actually pepper our code with logging stuff.

I think it's probably best for each person to go through and add logging to their own modules b/c you know what needs to be logged, but let's have this issue so we can point out areas that need to be logged as we notice them. 

The only basic guidelines i can think of are
* use `DEBUG` for repetetive/looping things -- stuff that you might want to have access during debugging, but not normal operation.
* use `INFO` to log all basic events like responses to interaction, settings changing, etc. It's good to double-log, eg. the UI sets a control in the controller, the UI logs it has done so and the controller has logged it has received the control -- each module should create its own logfile so it's not redundant information and is useful to diagnose problems.
* the alarm manager will always use `EXCEPTION` when raising alarms so alarms are always logged.

# vent.alarm

# vent.common

# vent.controller

# vent.coordinator

# vent.gui

- [ ] Initialization - debug log creation of each object, info log summary of monitored and controlled values
- [ ] starting/stopping
- [ ] control changed including means of change (typed, slider, plot)
- [ ] state changed (lock/unlock)
- [ ] received control value from controller
- [ ] received/dismissed/stopped showing alarm objects
- [ ] saving/loading persistent state

# vent.io

mschottdorf, 2020-09-05:
@sneakers-the-rat This is all done, isn't it?

mschottdorf, 2020-10-09:
done!

closed 2020-10-09 by mschottdorf