CohenLabPrinceton/pvp #79: Pressurized Gas Requirements (ISO 80601-2-80)

CohenLabPrinceton, 2020-06-02:
a) If the VENTILATORY SUPPORT EQUIPMENT is intended to be connected to a MEDICAL GAS PIPELINE SYSTEM complying with ISO 7396‐1, then it

- [ ] a1: Must comply with 80601 throughout RATED input range. 

- [ ] a2: shall not cause an unacceptable RISK under the SINGLE FAULT CONDITION of 1 000 kPa.

b) If VSE has max RATED input pressure in excess of 600 kPA, then the SFC is 1200 kPa.

- [ ] validate that we are below 600 kPa for rated input pressure.

CohenLabPrinceton, 2020-06-02:
Notes from ISO 80601:

_TL;DR = it's OK to have a regulator to keep the flow/pressure where we want. Also, even with overpressure we want flow to continue through the VBS and it's OK if flow rates are outside tolerance at this point._ 

NOTE 1 Internal pressure regulators can be needed to accommodate the SINGLE FAULT CONDITION of maximum input pressure, as well as the RATED range of input pressure.

NOTE 2 Under the SINGLE FAULT CONDITION of overpressure, it is desirable for gas to continue to flow to the VBS. Under this condition, the flowrate from the VENTILATORY SUPPORT EQUIPMENT is likely to be outside of its specification._

CohenLabPrinceton, 2020-06-02:
I've checked off (a1) because I think it just means that the VBS needs to function as described for all pressure within the RATED range we provide in our spec sheet.

zajdel, 2020-06-02:
For (b), our regulator is to be set to deliver 50 psi input pressure, which is 344 kPa < 600 kPa.

zajdel, 2020-06-02:
Concerning (a2), our regulator (SMC AR25-02BG, datasheet attached) has a maximum operating pressure of 1000 kPa and a proof pressure of 1500 kPa.

closed 2020-06-02 by zajdel