CohenLabPrinceton/pvp #80: Running as two processes on the PI: "Address already in use" OSError

mschottdorf, 2020-06-02:
I'm running 
```
    coordinator = get_coordinator(single_process=False, sim_mode=True)
    app, gui = launch_gui(coordinator)
```
on the Pi, and get the following error:
```
manuel@raspberrypi:~/Ventilator-Dev $ ipython ./sandbox/run_UI_pi.py
2020-06-02 10:28:26,924 - vent.common.prefs - INFO - Loaded prefs from /home/manuel/vent/prefs.json
2020-06-02 10:28:26,926 - vent.common.prefs - INFO - Saved prefs to /home/manuel/vent/prefs.json
2020-06-02 10:28:27,408 - vent.coordinator.coordinator - INFO - coordinator init
2020-06-02 10:28:27,427 - vent.coordinator.rpc - INFO - controller process init
2020-06-02 10:28:27,430 - vent.controller.control_module - INFO - controller init
Process Process-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/manuel/Ventilator-Dev/vent/coordinator/rpc.py", line 56, in rpc_server_main
    server = SimpleXMLRPCServer((addr, port), allow_none=True, logRequests=False)
  File "/usr/lib/python3.7/xmlrpc/server.py", line 605, in __init__
    socketserver.TCPServer.__init__(self, addr, requestHandler, bind_and_activate)
  File "/usr/lib/python3.7/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/usr/lib/python3.7/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 98] Address already in use
```
Consistent with that, the UI pops up, but no communication with the controller.

sneakers-the-rat, 2020-06-03:
i get this when a Python process is hung open from a previous run. we definitely should check for that and kill the process on startup. @sunnyszy it seems like you've got your hands full with the interview, would you like me to work on the coordinator re: this & other process management stuff?

mschottdorf, 2020-06-03:
Just an update, I still haven't managed to run the UI with two processes on the PI. Doing the optimization only with Hal & Controller rn

mschottdorf, 2020-06-05:
Update: there was an old python process listening on our default port (9533, specified in `rpc.py`). Found it with `lsof -i:9533`, killed it, and then the UI worked.

mschottdorf, 2020-06-05:
@sneakers-the-rat might be food for thought when rewriting the task-controller in the coodinator

closed 2020-08-09 by mschottdorf