CohenLabPrinceton/pvp #81: Pressure/Flow Compatibility Requirements (201.4.11.101.2)

CohenLabPrinceton, 2020-06-02:
If the VENTILATORY SUPPORT EQUIPMENT is intended to be connected to a MEDICAL GAS PIPELINE SYSTEM complying with ISO 7396‐1, then

- [ ] a) the RATED range of input pressure shall cover the range specified in ISO 7396‐1, and

 b) under NORMAL CONDITION,

- [ ] 1) the maximum 10 s average input flowrate required by the VENTILATORY SUPPORT EQUIPMENT for each gas shall not exceed 60 l/min at a pressure of 280 kPa, measured at the gas input port, and

- [ ] 2) the transient input flowrate shall not exceed 200 l/min averaged for 3 s, **OR..**.

- [ ] (OR) 3) the ACCOMPANYING DOCUMENTS shall disclose the following: 

- [ ] i) the maximum 10 s average input flowrate required by the VENTILATORY SUPPORT EQUIPMENT for each gas at a pressure of 280 kPa, measured at the gas input port;

- [ ] ii) the maximum transient input flowrate averaged for 3 s required by the VENTILATORY SUPPORT EQUIPMENT for each gas at a pressure of 280 kPa, measured at the gas input port;

- [ ] iii) a warning to the effect that this ventilator is a high‐flow device and should only be connected to a pipeline installation designed using a diversity factor that allows for the indicated high flowrate at a specified number of terminal outlets, in order to avoid exceeding the pipeline design flowrate, thereby minimizing the risk that the ventilator interferes with the operation of adjacent equipment.

CohenLabPrinceton, 2020-06-02:
I am a bit confused about the language "Average input flowrate *required* by the VSE..." Sounds like we need to show that our 10s average is <= 60 l/min at the gas input port. Seems like we'd address this with actual flow data that we'd compile into a figure.

CohenLabPrinceton, 2020-06-02:
We also need to establish a test protocol to collect these data and decide who will do it.

juliennelachance, 2020-06-16:
Yes, it sounds like we should just record pressure and flow logs over the course of 10+ cycles. The flow rate should be well within range.

CohenLabPrinceton, 2020-06-17:
@juliennelachance we need to fix the pressure at 280 as well. @zajdel can we get these data relatively soon since we don’t need alarms or anything fancy?

sneakers-the-rat, 2020-06-22:
dk if i'm needed on this issue, unassign me if no gui stuff needed <3

closed 2020-10-09 by CohenLabPrinceton