CohenLabPrinceton/pvp #82: Overstretch on the test-lung during remote testing

CohenLabPrinceton, 2020-06-02:
Test-lung over-stretched and damaged the springs and the lung bag itself. 

Context: Manuel was remotely testing control and noted unstable and surprising behaviors in the system; possibly stuck or erratic valves? 

Short-term Responses:

- [ ] Mechanical limiter to stop this happening again

- [ ] Swap to cheap test lung bag until we're confident

- [ ] Rush order springs

Long-term Responses--Manuel/Tom?

CohenLabPrinceton, 2020-06-02:
Tom ordered springs already.

zajdel, 2020-06-02:
We've swapped to the balloon bag already. I have also moved the pressure sensor the right location.

I'm testing the mechanical limiter that Julie spec'd some time ago, since it fits into the manifold we're currently using and testing that.

zajdel, 2020-06-02:
So the pressure relief valve was installed in the wrong direction. It doesn't easily fit into the manifold the way it's supposed to, so I put it in the wrong way. I would say we had a mechanical failure of pressure limiting that could have been avoided. Earlier versions of Manuel's code would terminate while leaving the inspiratory valve open. In this 50 psi system, that leads to instant inflation without a proper pressure cutoff. I'm going to make sure that I know how this new pressure valve needs to be installed.

mschottdorf, 2020-06-02:
I looked at the logs from this morning. The situation happened in conjunction of three failures: (1) inspiratory valve stayed open, while expiratory valve was closed, (2) pressure measurement was not configured properly; therefore HAPA was not triggered [from the logs: recorded pressure always <50cmH2O] and (3) hardware failure of the relief valve.

To avoid this happening in the future:

1. While the code implements high-airway-pressure-alert and management, it was not triggered [Currently set to 80 cmH2O, opens the expiratory valve, and closed the inspiratory valve]. Need to make sure that config files is the right one for the device. Possible tests could be: atmospheric pressure should be around 0. What else?

2. To test for sluggish valves, could `set_ex` and `set_in` multiple times. 

3. Make flowing the relief valve a mandatory test before playing with hardware.

zajdel, 2020-06-02:
The relief valve is for sure working now, so that's a backup safety that we have now, though that's up to 0.75 psi (50 cm H2O), so that should be the hardware-enforced limit.

CohenLabPrinceton, 2020-06-02:
@zajdel @mschottdorf thanks for tracking things down. I'm curious that the pressure never exceeded 50 cmH2O and I want to make sure this is real because I would have thought that enough pressure to overinflate that lung and overcome compliance would have gone higher than 50 cmH2O. Can we please confirm that the pressure sensor works properly; might have just been where it was positioned?

mschottdorf, 2020-06-02:
Let's do this tomorrow @zajdel; I've changed a few minor things in the code, but the pressure remains surprisingly low. Can we produce a defined pressure in the system, and I check what the controller believes the pressure to be?

zajdel, 2020-06-03:
I have a physical gauge I can attach in the system near the pressure sensor. That should help at least verify the pressure. Another way would be to remove the test lung and run at full tilt, letting the pressure relief valve set the system pressure to a known 50 cmH2O.

zajdel, 2020-06-04:
Test lung sent to IngMar Medical for repairs, and Manuel and I spent some time calibrating the pressure sensor in the airway, catching an important scaling error caused by our using a differential pressure sensor. This, along with safeties that Manuel put in the code, makes it so that this will not happen again!

closed 2020-06-04 by zajdel