CohenLabPrinceton/pvp #87: TestReport: Testing Error Reporting

ProfDJCohen, 2020-06-03:
Ref: CR501 201.5.101.3

- [ ] Updated?

VENTILATORY SUPPORT EQUIPMENT testing errors
--

- [ ] a) For the purposes of this document, declared tolerances shall include the measurement uncertainty.

- [ ] b) The MANUFACTURER shall disclose the measurement uncertainty of each disclosed tolerance in the technical description.

ProfDJCohen, 2020-06-03:
@zajdel @CommReteris @sneakers-the-rat @Sophie We need to make sure that we have a table of measurement uncertainty, etc. that we report in our labeling parameters and for every measured/derived parameter.

sneakers-the-rat, 2020-06-03:
should the UI display static +/- % error tolerance message next to measured parameters?

CohenLabPrinceton, 2020-06-03:
Nope. These need to go in the labeling documentation and disclosures in the manual, not the machine UI

CohenLabPrinceton, 2020-06-18:
Update: @zajdel will test this using testing conditions for Pressure Control table from the ISO table. @SophieDvali can help fit the data if needed.

SophieDvali, 2020-06-19:
I'm a little confused about the wording here, does this just mean we have to report the uncertainty of all measured quantities? Is this separate from the EUA items for reporting the error of the individual measured quantities?

CohenLabPrinceton, 2020-06-22:
@SophieDvali these are the same things. We need to report the error for any quantities we show the user. I think if you have the error values for flow and volume then we should be good.

sneakers-the-rat, 2020-06-22:
unassigning myself since this is a docs thing for the hardware team. lmk if you need help using the docs tho.

closed 2020-10-09 by CohenLabPrinceton