CohenLabPrinceton/pvp #88: TestReport: UI and Units of Measure

ProfDJCohen, 2020-06-03:
Ref 1: CR501 201.7.4.3 / 80601-2-80

Units of measure

a) All gas volume, flowrate and leakage specifications:

- [ ] 1) shall be expressed at STPD (STANDARD TEMPERATURE AND PRESSURE, DRY);

- [ ] 2) except those associated with the VBS which shall be expressed at BTPS (BODY

TEMPERATURE AND PRESSURE, SATURATED).

- [ ] bb) The unit of AIRWAY PRESSURE measurement shall be capable of being configured to be expressed in hPa

ProfDJCohen, 2020-06-03:
@sneakers-the-rat we need a 'units' setting where we can switch to hPa. I am not sure about the reporting of BTPS since we're not measuring anything that would be affected by that (I think).

sneakers-the-rat, 2020-06-20:
changing value over 2 orders of magnitude is actually sorta tricky when ur space constrained. got most of the way there but need to quit for the night.

sneakers-the-rat, 2020-06-21:
https://github.com/CohenLabPrinceton/Ventilator-Dev/commit/4c7c929df47941c2f3d7aad93c31c2311e44b339

sneakers-the-rat, 2020-06-21:
surprisingly hard to do

closed 2020-06-21 by sneakers-the-rat