CohenLabPrinceton/pvp #89: TestReport: Certification and Spec'ing of Single Use Parts

ProfDJCohen, 2020-06-03:
- [ ] a) if the VENTILATORY SUPPORT EQUIPMENT, its parts or ACCESSORIES are intended for single use,

information on characteristics and technical factors known to the MANUFACTURER that could pose a
RISK if the VENTILATORY SUPPORT EQUIPMENT, its parts or ACCESSORIES were reused.

b) if the VENTILATORY SUPPORT EQUIPMENT, its parts or ACCESSORIES are intended for single use,
information regarding:

- [ ] 1) the intended duration of use; and

- [ ] 2) the consequences for the PATIENT if use is beyond the specified duration of use.

ProfDJCohen, 2020-06-03:
@julie @zajdel all of our disposable parts for single-use need spec-sheets and justifications for duration of use/issues if not complied with. This is really important for the documentation I think. We'll need a complete list of all EXTERNAL accessories, biocompatibility of these, and certification info.

zajdel, 2020-06-16:
List of single-use parts:
- PEEP valve
- HMEF
- DAR filters
- Respiratory tubing

CohenLabPrinceton, 2020-06-18:
@SophieDvali maybe can help get the spec sheets for these. 

Where does this go in the test report? Manual?

closed 2020-10-09 by CohenLabPrinceton