CohenLabPrinceton/pvp #93: TestReport: Oxygen combustion safety

ProfDJCohen, 2020-06-03:
Ref: CR501 Clause 11 / IEC 60601-1 (11.2.2.1 is MISSING)

**RISK of fire in an OXYGEN RICH ENVIRONMENT**

If the medical device is intended to be used with oxygen:

- [ ] 	Describe means of protection to reduce the risk of ignition

If the medical device is intended to be used with oxygen operating at medical gas pipeline pressures:

- [ ] Describe means of protection to taken to ensure that auto-ignition from adiabatic compression cannot occur

ProfDJCohen, 2020-06-03:
AAMI CR502 implies that auto-ignition is only a risk at 5 bar or higher. 50 psig is 3.5 bar, so I think we're OK there. However, we need careful discussion of combustion risk mitigation.

juliennelachance, 2020-06-03:
Ignition risk mitigation includes: no contactors (like relays: everything we're using is solid-state) so as to avoid arcs; enclosure of the electronics; avoiding aluminum in the oxygen pathway; etc.

CohenLabPrinceton, 2020-06-03:
@juliennelachance thanks, but we need to be exhaustive, so please do not say ‘etc’. If there is something else that helps, write it explicitly. A concise, bulleted list is best.

CohenLabPrinceton, 2020-06-18:
@zajdel speaker could be an issue.

closed 2020-10-09 by CohenLabPrinceton