CohenLabPrinceton/pvp #95: TestReport: Cleaning and Disinfection

ProfDJCohen, 2020-06-03:
CLEANING and DISINFECTION of ME EQUIPMENT or ME SYSTEM

Describe the cleaning/disinfection protocol of the medical device and accessories

See ISO 80601-2-80 snapshot below (couldn't copy-paste)
[image]

ProfDJCohen, 2020-06-03:
We need to discuss how to clean the inside of the vent and any 3D printed parts. We can discuss in our risk management assessment that we use dual HEPA filters specifically to ensure that anything leaving the vent is sterile so nothing can reach the patient that isn't sterile UNLESS there is a problem in the respiratory circuit itself, which is single-use. Thoughts?

CohenLabPrinceton, 2020-06-18:
Need @DrDan opinion on this. Likely household ammonia-based cleaning following a standard protocol.
What about the 3D printed manifolds? How do we clean PLA/ABS/PETG. 
@SophieDvali can you look into the solenoid valves and the 3D printed manifolds? DrDan should be able to help recommend cleaning solutions.

SophieDvali, 2020-06-22:
There's a bunch of Czech labs that tested all sorts of cleaning methods for 3D printed face shields (some of the tests were done specifically on COVID-19) they include things like UV-C, Isopropanol and Hydrogen peroxide (the full list is here: https://help.prusa3d.com/en/article/prusa-face-shield-disinfection_125457/ ) This is for PETG specifically. Since this is for face shields some of these might not work for our case.

closed 2020-10-09 by CohenLabPrinceton