CohenLabPrinceton/pvp #97: TestReport: Alternate power supply

CohenLabPrinceton, 2020-06-03:
Additional requirements Interruption of the power supply/SUPPLY MAINS   to ME EQUIPMENT

Describe:

- [ ] •Means used to provide alternate power source

- [ ] •Means to determine the state of the alternate power source

- [ ] •Run time of alternate power source 

- [ ] •Alarm indications

CohenLabPrinceton, 2020-06-03:
@sneakers-the-rat we may need to incorporate an alarm for this in the UI that detects IF the alternate supply is in use. @zajdel easiest way to do this for an arbitrary UPS power supply? Alternately, we can spec one that actually alarms us when it is in use using an audible alarm.

sneakers-the-rat, 2020-06-03:
I have a battery pack, but would be useful to have the battery pack that we plan on using for the final thing so i can test code that checks on power source

CohenLabPrinceton, 2020-06-03:
@sneakers-the-rat I think we need an actual UPS that can keep the whole machine running for this. It might be an optional standard so we could just say 'not included', but it's better to have it. The Pi battery is also important, and maybe that would only engage when the UPS is needed. @juliennelachance do you and Rengo have a favorite battery pack we were going to use? Otherwise I'd say just use a UPS because battery packs are fire risks.

zajdel, 2020-06-03:
Is the UPS to be included within the project enclosure, or can it be external?

CohenLabPrinceton, 2020-06-04:
External ok for EUA @zajdel

zajdel, 2020-06-05:
The UPS we use in the lab (and many others, it seems) comes with a USB communication port. I am not knowledgeable on this subject, but this USB port is designed to be used with UPS monitoring software. I will look into what we would need to do to poll the UPS and get a status update.

Here's a tutorial about doing just that with a Pi.
[](https://medium.com/@mike.reider/monitor-cyberpower-ups-devices-with-raspberry-pi-99559725dbb8)

Looks like the NUT (Network UPS Tools) library, or something like it, will need to be installed.

sneakers-the-rat, 2020-06-05:
lmk whatever ups you have/how to query it and i can add status/timer in ui

CohenLabPrinceton, 2020-06-07:
If the UPS has simple LEDs that light to indicate charge/function, that's also fine. COnsidering that a UPS is optional for the EUA, I'd say spend minimal time here. @sneakers-the-rat @zajdel

zajdel, 2020-06-09:
Our UPS doesn't have an LED, but its LCD shows if it's running off of the battery or not.

We will use an [OMNI1500LCDT](https://www.tripplite.com/omnismart-lcd-120v-1500va-810w-line-interactive-ups-tower-lcd-display-usb-port-energy-star~OMNI1500LCDT) which is rated 810W output which should cover our system.

To determine runtime on battery, I would get a power monitor (like a Kill a Watt) and monitor the vent while it runs at 20 breaths/minute to estimate power consumption.  Then we'd compare that against the capacity of the battery, which is cited as 7.5 minutes at half load (405 Watts). Our power use is likely <50 W, so I would estimate on the order of an hour of battery power. Still, better measure the power output to be certain.

closed 2020-10-09 by CohenLabPrinceton