CohenLabPrinceton/pvp #98: TestReport: Accuracy of Controls and Visibility in the UI

CohenLabPrinceton, 2020-06-03:
This is all Jonny: 

Summary of IEC 60601-1 Clause 12:

- [ ] Controls MAY be allowed to vanish to free up screen space either automatically on operator command

- [ ] if controls vanish, they need to reappear during any alarm condition

- [ ] All controls and indicators must be 'clearly legible'. See screenshot

- [ ] Lux range ideally is 100 lx to 10,000 lx. (This may not be necessary, but let's see what our displays can handle). 

[image]

CohenLabPrinceton, 2020-06-03:
@juliennelachance here because of screen discussion.

CohenLabPrinceton, 2020-06-03:
nm, not required via AAMI update

CohenLabPrinceton, 2020-06-03:
We need to agree on some quick bullet point responses to this here, I think.

sneakers-the-rat, 2020-06-03:
I have a battery pack, but would be useful to have the battery pack that we plan on using for the final thing so i can test code that checks on power source

sneakers-the-rat, 2020-06-22:
prompting for unsafe values implemented here: https://github.com/CohenLabPrinceton/Ventilator-Dev/commit/eccf78b56d75e1a997ce4f932677dab76ad6e7d0

- each controlled value has a `safe_range` .
- if the user attempts to set a value outside of the `safe_range`, they are prompted with a message like this:

[image]

- if the user presses 'ok,' the value is set and the user is not prompted until the value returns to & exits the safe range again.
- if the user presses 'cancel,' the value is set to the most recent value.

sneakers-the-rat, 2020-06-22:
locking impemented: https://github.com/CohenLabPrinceton/Ventilator-Dev/commit/f93c78bc3ba32311b79aa461a9304ae8db18e3ca

before starting:
[image]

after starting, the ui starts in 'locked' mode by default, indicated by the green 'lock' icon and color-corresponding border around controls and the recession of the control subboxes into the main box background. While locked, none of the controls can be changed. 

[image]

The user can unlock the ui, allowing controls to be changed, by clicking the lock.

[image]

closed 2020-08-12 by sneakers-the-rat