CohenLabPrinceton/pvp #99: TestReport: Pressure Controlled Vent standard

CohenLabPrinceton, 2020-06-03:
This is a big one. I've uploaded direct from the ISO standard for this. This needs a real discussion. Someone besides me should go through and parse each instruction here as a checkbox. 

[ISO 80601-2-80-2018_PCBreath.pdf](https://github.com/CohenLabPrinceton/Ventilator-Dev/files/4726153/ISO.80601-2-80-2018_PCBreath.pdf)

CohenLabPrinceton, 2020-06-03:
@mschottdorf @juliennelachance @zajdel we need to have a meeting go through this I think and design the test approach and make sure it's rock-solid so we don't have to repeat it and ensure we log data, etc.

zajdel, 2020-06-03:
I see why we have the QuickLung. That takes care of issues 7-10 in the test set up.

[image]

Checklist for required test components:
- [ ]  1. Ventilatory support equipment under test (that would be the Dinky)
- [ ]  2. Pressure sensor (proximal) - we need to finalize the connector/port we are using.
- [ ]  3. Flow sensor - sensiron probably works - the rise time requirement only applies for volume-controlled breath types
- [ ]  4. Artificial leakage - Not sure what this requires yet.
- [ ] 5. Data acquisition system - minimum sample rate of 200 samples/s (do we achieve this?)
- [ ] 6. Temperature sensor - We have one but I have not yet set it up. Should communicate via SPI with the main board. We have connectors for this.
- [ ] 7. Series resistance with test lung (QuickLung adjustable)
- [ ] 8. Pressure sensor with a 10% to 90% rise time of no greater than 10 ms (double-check this for our 1psi 4v mini - Quicklung port)
- [ ] 9. compliance of test lung (QuickLung adjustable)
- [ ] 10. test lung (QuickLung)

zajdel, 2020-06-04:
Table 201.105 from ISO.80601-2-80-2018

| Test number | Intended delivered volume (mL) | Compliance (mL (hPa)^-1) | Linear resistance (hPa(L/s)^-1) +/- 10% | Leakage (mL/min) +/- 10% | Ventilatory frequency (breaths/min) | Inspiratory time (s) | Pressure (hPa) | PEEP (hPa) |
|-------------|--------------------------------|--------------------------|-----------------------------------------|--------------------------|-------------------------------------|----------------------|----------------|------------|
| 1           | 500                            | 50                       | 5                                       | 0                        | 20                                  | 1                    | 10             | 5          |
| 2           | 500                            | 50                       | 20                                      | 0                        | 12                                  | 1                    | 15             | 10         |
| 3           | 500                            | 20                       | 5                                       | 0                        | 20                                  | 1                    | 25             | 5          |
| 4           | 500                            | 20                       | 20                                      | 0                        | 20                                  | 1                    | 25             | 10         |
| 5           | 500                            | 50                       | 5                                       | 5000                     | 20                                  | 1                    | 25             | 5          |
| 6           | 500                            | 50                       | 20                                      | 10000                    | 12                                  | 1                    | 25             | 10         |
| 7           | 300                            | 20                       | 20                                      | 0                        | 20                                  | 1                    | 15             | 5          |
| 8           | 300                            | 20                       | 50                                      | 0                        | 12                                  | 1                    | 25             | 10         |
| 9           | 300                            | 10                       | 50                                      | 0                        | 20                                  | 1                    | 30             | 5          |
| 10          | 300                            | 20                       | 20                                      | 3000                     | 20                                  | 1                    | 25             | 5          |
| 11          | 300                            | 20                       | 50                                      | 6000                     | 12                                  | 1                    | 25             | 10         |
| 12          | 200                            | 10                       | 20                                      | 0                        | 20                                  | 1                    | 25             | 10         |

Notes:
- Volume in column 2 is intended to be used for selection of the test conditions and parameters based on the intended delivered volume of Ventilator Support Equipment (what does this mean?)
- For the purpose of this test, VSB is set up with artificial leakage (column 4) at a constant pressure of 20 hPa
- In the case that end-expiratory flow does not reach zero, reduce the ventilatory frequency (column 6) until it does for at least 50 ms
- The rise time of the ventilatory support equipment should be set to a value that ensures that the set pressure can be reached within the inspiratory time (i.e. 1 second in every case - your time to shine, Manuel!)
- For the purposes of this test, the set pressure is relative to set PEEP (that means =  PEEP + pressure? i.e. test number 1, we're looking for 15 hPa in our pressure readings?)

zajdel, 2020-06-04:
Section on pressure-controlled breath type (201.12.1.102)
- [ ]  a) If pressure-controlled breath type is provided, then with a pressure-controlled breath type selected and VSE operating in normal condition, the accuracy as determined for the test settings and conditoins specified shall be disclosed as the maximum bias error and maximum linearity error (example: +/- 3.0 + (5% of the set pressure)) hPa)
- [ ] b) This disclosure includes at least
       1) maximum error of airway pressure (Paw) at end of inspiratory phase in relation to set value
       2) maximum error of airway pressure (Paw) at the end of the inspiratory phase in relation to the set value under leak condition
       3) maximum error of airway pressure (Paw) at end of expiratory phase in relation to set value
       4) maximum error of airway pressure (Paw) at end of expiratory phase in relation to set value under leak condition
- [ ] c) All of the errors may be separately reported for the following ranges of intended delivered volume:
       1) Vdel >= 300 mL
       2) 300 mL >= Vdel >= 50 mL
- [ ] d) The accuracy of the performance of the ventilatory support equipment shall either be:
       1) determined for each VBS configuration indicated in the instructions for use; or
       2) determined for the worst-case VBS configuration indicated in the instructions for use (NOTE 1: The worst-case VBS configuration can be different for each error or each nominal delivered volume range)
- [ ] e) If worst-case VBS configurations are used, the rationale for their selection shall be documented in the *risk management file*

_Check compliance by inspection of the *risk management file* for the rationale, if applicable, and with the following tests for end-inspiratory and end-expiratory pressure errors:_
- [ ] f) Set up the ventilatory support equipment as shown in Figure 201.101 (above)
- [ ] g) If applicable, determine or input the VBS compliance required for compliance correction as indicated in the instructions for use and activate this correction. If a humidifier is used, fill the humidifier to the maximum water level prior to determining the VBS compliance.
- [ ] h) Utilize the rest parameters of the first applicable row (selected by typical intended delivered volume) of Table 201.105. Wait until steady-state conditions are achieved.
- [ ] i) Determine the airway pressure (Paw) at the end of the inspiratory phase as the average over the preceding 50 ms
- [ ] j) Compare the result with the pressure setting for the test. Confirm that the resulting difference is within the tolerance indicated in the instructions for use.
- [ ] k) Determine the PEEP as the averages of the airway pressure measurements (Paw) over the last 50 ms of the expiratory phase
- [ ] l) Compare the result with the PEEP setting for the test. Confirm that the resulting difference is within the tolerance indicated in the instructions for use.
- [ ] m) Repeat g) to l) for 30 consecutive breaths
- [ ] n) Repeat g) to m) for each applicable row (selected by intended delivered volume) of Table 201.105
- [ ] o) If a humidifier is included in the VBS, repeat the airway pressure tests with the minimum humidifier water level without re-determining the VBS compliance
- [ ] p) If the ventilatory support equipment permits operation without compliance correction, repeat g) to o) without compliance correction.
- [ ] q) Compare each result to the tolerance indicated in the instructions for use.

zajdel, 2020-06-04:
Main takeaways for now:
- Instructions for use keep popping up. Our test results will have to corroborate with the instructions for use.
- We seem to need to disclose 4 maximum errors, all for the airway pressure: Inspiratory phase, Inspiratory phase + leak, Expiratory phase, Expiratory phase + leak
- Are we allowing operation without VSB compliance correction? Seems we are required to allow operation with VSB compliance correction.
- What is our intended delivered volume for the ventilatory support equipment? Would we pick 500 mL (for adults?) and then go with tests 1-6? Are tests 7-12 applicable?
- We need to determine our steady state airway pressure, then average for 50 ms to get our test results both in PEEP and PIP phases. And we must do this for 30 cycles for each test condition.
- If we're using a humidifier, we have to do all tests twice: once at maximum humidifier water level and once at minimum humidifier water level.

CohenLabPrinceton, 2020-06-04:
@zajdel we do NOT need a temperature sensor for formal logging. That's actually excluded somewhere else in the EUA as irrelevant if we don't use a heated humidifier.

zajdel, 2020-06-04:
> 
> 
> @zajdel we do NOT need a temperature sensor for formal logging. That's actually excluded somewhere else in the EUA as irrelevant if we don't use a heated humidifier.

Since we are not using a heated humidifier, (g) part ii and (o) are not relevant. Also, we will not need to use a temperature sensor (component 6). These parts have been checked off.

Is the data acquisition system for test (component 5) to be separate/distinct from the ventilator itself (component 1)?

zajdel, 2020-06-04:
For component 8 in the test setup, pressure sensor, our Amphenol sensors (https://www.mouser.com/datasheet/2/18/1/DS-0103-Rev-A-1499253.pdf) have a response time to get to 90% pressure applied of < 100 microseconds.

juliennelachance, 2020-06-04:
The current read time is 2 ms per sensor. (So if you need to read more than one sensor, of course, it adds to the total time.)

juliennelachance, 2020-06-04:
Also: I am in favor of using the HME-F luer lock connector for now. So that takes care of checkboxes 2 and 5

closed 2020-10-06 by zajdel