CohenLabPrinceton/pvp #101: TestReport: Airway Pressure Protection

CohenLabPrinceton, 2020-06-04:
Ref: ISO 80601-2-80 / 201.12.4.101.1

**General**

- [ ] a) The VENTILATORY SUPPORT EQUIPMENT shall be equipped with MONITORING EQUIPMENT to indicate the AIRWAY PRESSURE.

- [ ] b) The site of actual measurement may be anywhere in the VENTILATOR BREATHING SYSTEM, but the indicated value shall be referenced to the PATIENT‐CONNECTION PORT.

- [ ] c) Under steady‐state conditions, the indicated AIRWAY PRESSURE shall be accurate to within ± (2 + 4 % of the actual reading) hPa (cmH2O).

CohenLabPrinceton, 2020-06-04:
We need to confirm the error/accuracy of the pressure sensor. Do we have this in the spec sheet? How else should we quantify this?

juliennelachance, 2020-06-04:
The maximum linearity/hysteresis error at standard temperature is reported as +/- 0.5 %fs. Corresponding to a full-scale of 1 psi, this gives us an error of around 0.35 cm H20. 

Datasheet for reference: 
https://www.mouser.com/datasheet/2/18/1/DS-0103-Rev-A-1499253.pdf
We use the 1 PSI-D-4V-MINI as our system pressure sensor (with one side open to ambient).

zajdel, 2020-06-04:
Just to add to this, the nominal output voltage span is +/- 1.9 V to +/- 2.1 V. This could result in an error of up to 0.1 V / 2 V range (assumed) -> 0.05 psi or 3.5 cm H2O, worst case (5% off measured). We can easily calibrate the offset voltage (and should!), but may consider including an additional calibration procedure to measure the output span if this potential error must be reduced.

zajdel, 2020-06-09:
Our maximum error of +/- 3.5 hPa at full scale (1 psi = 70 hPa) is within the ± (2 + 4 % of the actual reading) hPa (cmH2O) requirement, even with the additional linearity error of 0.35 hPa that Julie cites, for a total error of +/- 3.85 hPa at full scale (our worst case). At full scale (70 hPa), we are allowed +/- 4.8 hPa error, according to the requirements.

I have checked the final box so this issue is pending closure.

CohenLabPrinceton, 2020-06-18:
@juliennelachance @zajdel please confirm the error statement.

closed 2020-10-09 by CohenLabPrinceton