CohenLabPrinceton/pvp #102: TestReport: Low Airway Pressure Alarm

CohenLabPrinceton, 2020-06-04:
Low AIRWAY PRESSURE ALARM CONDITION

- [ ] a) The AIRWAY PRESSURE MONITORING EQUIPMENT shall be equipped with an ALARM SYSTEM that detects an

ALARM CONDITION to indicate when the low AIRWAY PRESSURE ALARM LIMIT is reached.

b) The low AIRWAY PRESSURE ALARM CONDITION

- [ ] 1) shall be at least a MEDIUM PRIORITY, unless

- [ ] 2) an INTELLIGENT ALARM SYSTEM, based on additional information, determines that

- [ ] i) the low AIRWAY PRESSURE ALARM CONDITION is suppressed, or

- [ ] ii) its priority is changed, or

- [ ] 3) may start at LOW PRIORITY, and

- [ ] 4) if this state continues, escalate to MEDIUM PRIORITY.

- [ ] c) The low AIRWAY PRESSURE ALARM SIGNAL may be inactivated with ALARM OFF.

- [ ] d) ALARM OFF may be activated by the VENTILATORY SUPPORT EQUIPMENT.

- [ ] e) The low AIRWAY PRESSURE ALARM LIMIT may be

- [ ] 1) pre‐adjusted,

- [ ] 2) RESPONSIBLE ORGANIZATION‐adjustable,

- [ ] 3) OPERATOR‐adjustable,

- [ ] 4) VENTILATORY SUPPORT EQUIPMENT‐adjustable, or

- [ ] 5) a combination of OPERATOR‐adjustable and VENTILATORY SUPPORT EQUIPMENT‐adjustable.

- [ ] f) If the AIRWAY PRESSURE ALARM LIMIT is adjustable by the VENTILATORY SUPPORT EQUIPMENT, a summary

description of the algorithm that determines the ALARM LIMIT value shall be disclosed in the instructions for use.

NOTE Depending on the type of ventilation mode being utilized, there can be more than one active ALARM LIMIT

CohenLabPrinceton, 2020-06-04:
@sneakers-the-rat @mschottdorf Here's alarm 1--LAPA. We discussed all of this, but please make sure we're still compliant and check the appropriate boxes. 

Please also take a stab at writing a brief blurb for how the algorithm determines the ALARM LIMIT (and how the operator might be able to adjust).

CohenLabPrinceton, 2020-06-18:
@sneakers-the-rat @mschottdorf please check things off

sneakers-the-rat, 2020-06-21:
https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/296b47f3baf5d0d1179008bd833855163a06210d/vent/alarm/__init__.py#L36

Low pressure alarm is raised as a LOW severity alarm when measured PIP for a given breath cycle is 10% lower than the set PIP. A Low pressure alarm is elevated to a MEDIUM severity alarm if measured PIP is below 15% below the set PIP and a LOW severity low pressure alarm has been active for two breath cycles.  The alarm can be dismissed, and if dismissed will not be re-raised until measured PIP returns above the LOW severity alarm threshold (-10% of set PIP).

sneakers-the-rat, 2020-06-21:
dk what ventilatory support equipment means

CohenLabPrinceton, 2020-06-22:
@sneakers-the-rat please confirm if the LAPA can be suppressed. I think everything else is good to go. VSE is the machine itself, I think, so that means if the machine calculates the alarm/severity, then describe how (which you did). Thanks!

sneakers-the-rat, 2020-06-22:
confirm LAPA can be suppressed - `latch = False` means if alarm conditions are no longer met the alarm is dismissed automatically, and if user requests dismissal the alarm is dismissed until pressure returns to and then exit normal range.
https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/296b47f3baf5d0d1179008bd833855163a06210d/vent/alarm/__init__.py#L38

closed 2020-10-09 by CohenLabPrinceton