CohenLabPrinceton/pvp #103: TestReport: High-Airway Pressure Alarm -- the BIG ONE

CohenLabPrinceton, 2020-06-04:
- [ ] a) If the MAXIMUM LIMITED PRESSURE is greater than 40 hPa (40 cmH2O), the VENTILATORY SUPPORT EQUIPMENT shall be equipped with an ALARM SYSTEM that detects a high AIRWAY PRESSURE ALARMCONDITION to indicate when the high AIRWAY PRESSURE ALARM LIMIT is reached.

 b) The high AIRWAY PRESSURE ALARM CONDITION

- [ ] 1) shall be HIGH PRIORITY, unless

- [ ] 2) an INTELLIGENT ALARM SYSTEM, based on additional information, determines that

- [ ] i) the HIGH AIRWAY PRESSURE ALARM CONDITION is suppressed, or

- [ ] ii) its priority is changed.

c) The high AIRWAY PRESSURE ALARM LIMIT may be

- [ ] 1) independently adjustable,

- [ ] 2) connected to an adjustable pressure limitation, or

3) related to the set pressure of the VENTILATORY SUPPORT EQUIPMENT.

- [ ] d) If the high AIRWAY PRESSURE ALARM LIMIT is independently adjustable, it shall not be possible to set the ALARM LIMIT to a value less than that of the adjustable pressure limitation.

- [ ] e) PATIENT‐generated transient pressure increases should not cause the high AIRWAY PRESSURE ALARM CONDITION. EXAMPLE Transient pressure increase caused by the PATIENT coughing.

- [ ] f) The high AIRWAY PRESSURE ALARM CONDITION DELAY shall not exceed 200 ms and the VENTILATOR shall

- [ ] 1) act to attempt to cause the pressure to start to decline within that duration, and

- [ ] 2) act to prevent the pressure from continuing to rise.

g) Whenever the AIRWAY PRESSURE ALARM CONDITION occurs, the VENTILATOR shall, within no more than
two breath cycles or 15 s, whichever is less, reduce the AIRWAY PRESSURE to either:

- [ ] 1) atmospheric pressure, or

- [ ] 2) the set PEEP level.

- [ ] h) During SINGLE FAULT CONDITION, the AIRWAY PRESSURE may be reduced below the set PEEP level.

CohenLabPrinceton, 2020-06-04:
@sneakers-the-rat @mschottdorf Here's the BIG ONE. Let's make sure we're all on the same page here. Lemme know if we should meet to go over these. We also need test protocols to make sure these work.

sneakers-the-rat, 2020-06-22:
https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/4c7c929df47941c2f3d7aad93c31c2311e44b339/vent/alarm/__init__.py#L77

a) does this say that the alarm should only fire when PIP is set above 40? currently alarm fires whenever measured PIP is 15% above set PIP.

b) HAPA is always HIGH, and remains actively displayed until 
* the measured pip is below 15% above set pip 
AND
* the user manually acknowledges & dismisses the alarm. 

c) (3) - HAPA threshold is computed as 15% above set PIP

d) alarm threshold not independently adjustable

e) @mschottdorf is the controller doing some filtering on the PIP values you're sending up? It looks like an alarm gets created regardless of whether or not the cough duration is met, i think you might need to make a `tentative_hapa` alarm object and check on its start time before creating the actual `self.HAPA` that's gotten by `get_alarms`

f) controller creates HAPA as soon as out-of-bounds pressure is detected, GUI polls alarms continuously from controller and raises them as soon as it gets them. 

f1 + f2) @mschottdorf can comment on this. I think it should probably be changed to g2 (set PIP to PEEP) : https://github.com/CohenLabPrinceton/Ventilator-Dev/blob/296b47f3baf5d0d1179008bd833855163a06210d/vent/controller/control_module.py#L398

closed 2020-10-09 by CohenLabPrinceton