CohenLabPrinceton/pvp #104: TestReport: Measurement of expired volume

CohenLabPrinceton, 2020-06-04:
- [ ] If VENTILATORY SUPPORT EQUIPMENT is equipped with MONITORING EQUIPMENT for indicating the volume expired through the PATIENT‐CONNECTION PORT, the accuracy shall be disclosed in the instructions for use.

We need to quantify the tidal volume accuracy. @SophieDvali ?

juliennelachance, 2020-06-04:
This is a good question for Sophie- who has the D-Lite data.

SophieDvali, 2020-06-04:
I can calculate this using the flow data from the mfc. This is asking for just the error in the flow right since the tidal volume is just flow times the expiration time?

SophieDvali, 2020-06-10:
When calculating just the error in the flow (from the dlite) I get a fractional error of around 5% which is reasonable and also consistent with what Philippe gets for the physics flow sensor. However, if I use error prop to get the error in the VT it becomes much higher (like 70%). I feel like there might be something wrong here.

closed 2020-10-09 by CohenLabPrinceton