CohenLabPrinceton/pvp #105: TestReport: Max Limited Pressure Protection Device

CohenLabPrinceton, 2020-06-04:
MAXIMUM LIMITED PRESSURE PROTECTION DEVICE

- [ ] a) A PROTECTION DEVICE shall be provided to prevent the AIRWAY PRESSURE from exceeding the MAXIMUM LIMITED PRESSURE under both NORMAL CONDITION and SINGLE FAULT CONDITION.

- [ ] b) The MAXIMUM LIMITED PRESSURE shall not exceed 90 hPa (90 cmH2O)

@juliennelachance @zajdel @sneakers-the-rat let's confirm our max pressure limit in the software as well as the specs on our safety valves. Please disclose all relevant specs in the comments below @juliennelachance @zajdel

zajdel, 2020-06-04:
We have a physical pressure release valve that will limit pressure to ~69 hPa (1 psi), placed just at the output of the inspiratory valve. The valve specs: [](https://www.mcmaster.com/4277T52)

Stated accuracy of set pressure release is +/- 20%, so within tolerance, the pressure limit is 55-83 hPa.

@mschottdorf can confirm his implementation of a software-based max pressure limit.

mschottdorf, 2020-06-05:
There is a software-coded maximum pressure limit, yes, that overwrites controller instructions and opens the exp. valve, and closed the inspiratory valve.
 If the hardware-config-file is correct, it actually works quite well :-)

zajdel, 2020-06-05:
It sounds like we have this covered, but I would like clarification on what the "single fault condition" is. We do limit pressure in two redundant ways during "normal operation." Does "single fault" simply mean that one component is not operating as desired?

CohenLabPrinceton, 2020-06-12:
@zajdel single fault condition means if if a key component fails (just one).

closed 2020-06-12 by CohenLabPrinceton