CohenLabPrinceton/pvp #109: TestReport: Hypoventilation Alarm

CohenLabPrinceton, 2020-06-12:
- [ ] a. VSE must monitor and alarm for hypoventilation

- [ ] b. Alarm OFF is allowable here

- [ ] c. NOTE The hypoventilation ALARM CONDITION can be determined, inter alia, by the measurement of the variations of AIRWAY PRESSURE, or expired volume, but possibly needs additional detection means. The hypoventilation ALARM CONDITION can also be determined by an INTELLIGENT ALARM SYSTEM utilizing one or more variables.

CohenLabPrinceton, 2020-06-12:
@mschottdorf @zajdel @sneakers-the-rat We need to describe how we do this.

sneakers-the-rat, 2020-06-30:
VTE can now set High/low VTE alarm by recording recent values, just need to know what i should use as the alarm threshold (eg. +/- 10% of set value like PIP, etc.)
https://github.com/CohenLabPrinceton/Ventilator-Dev/commit/88faef7bc1e5bf982c6abc3a2d71313329c64d1f

closed 2020-10-09 by CohenLabPrinceton