CohenLabPrinceton/pvp #110: TestReport: High Leakage Alarm

CohenLabPrinceton, 2020-06-12:
- [ ] a) The VENTILATORY SUPPORT EQUIPMENT may be equipped with an ALARM SYSTEM that detects a TECHNICALALARM CONDITION to indicate when conditions in the VBS reach the ALARM LIMIT for high leakage.

- [ ] b) If provided, the high leakage TECHNICAL ALARM CONDITION shall be at least MEDIUM PRIORITY, unless an INTELLIGENT ALARM SYSTEM, based on additional information, determines that the high leakage TECHNICAL ALARM CONDITION is suppressed, or its priority is changed

CohenLabPrinceton, 2020-06-12:
@mschottdorf @sneakers-the-rat need to confirm leakage alarm exists and works.

sneakers-the-rat, 2020-06-22:
Tell me how to parameterize leak and i'll implement in alarm manager.

CohenLabPrinceton, 2020-06-22:
@mschottdorf @zajdel we need to think about how the controller knows when there is a leak. Normally that would be handled by mass conservation with dual flow sensors, but we need an alternative. Might be issues with rise time, or maintaining pressure, etc.

closed 2020-10-09 by CohenLabPrinceton