CohenLabPrinceton/pvp #112: TestReport: Prevention of accidental changes

CohenLabPrinceton, 2020-06-12:
Protection against accidental adjustments

- [ ] a) Means of protection against accidental adjustment of controls that can create a HAZARDOUS SITUATION

shall be provided, including at least the following.

- [ ] 1) Switching on the VENTILATORY SUPPORT EQUIPMENT and starting the ventilation shall require two different actions.

- [ ] 2) It shall be possible to set all VENTILATORY SUPPORT EQUIPMENT parameters prior to starting

- [ ] 3) Turning off the ventilation shall require at least a sequence of two very deliberate actions.

- [ ] 4) Turning the VENTILATORY SUPPORT EQUIPMENT off shall require two different actions.

- [ ] 5) Setting the MAXIMUM WORKING PRESSURE greater than 40 hPa shall require at least a sequence of two very deliberate actions.

- [ ] b) These means may be accomplished by

1) utilizing hardware or software or a combination of both, or
2) two or more dedicated confirmation actions.
c) The USABILITY of these means of protection shall be evaluated in the USABILITY ENGINEERING PROCESS

CohenLabPrinceton, 2020-06-12:
@sneakers-the-rat @zajdel @juliennelachance big one on basic safety. I think our regulators, etc. are difficult to change accidentally. Our software controls include multiple confirmations, etc. Let's formally confirm these please by answering these check boxes.

CohenLabPrinceton, 2020-06-18:
@sneakers-the-rat please check off the boxes here.

sneakers-the-rat, 2020-06-22:
need to implement lock mode, on my life

sneakers-the-rat, 2020-06-24:
lock mode implemented

sneakers-the-rat, 2020-06-29:
5) is here: https://github.com/CohenLabPrinceton/Ventilator-Dev/commit/67745423707d28f3d8b9238433542be21451ac37

sneakers-the-rat, 2020-06-29:
what's the difference between (3) and (4)?

sneakers-the-rat, 2020-06-30:
ventilator can be turned off with two deliberate actions - pressing STOP and then pressing "OK" in dialog box that asks user to confirm stopping ventilation and warns that doing so prematurely is dangerous. if 3 and 4 are equivalent this is finished:

https://github.com/CohenLabPrinceton/Ventilator-Dev/commit/9302c9f0ef627b171b291b6ab04d4a5a42924289

sneakers-the-rat, 2020-08-12:
closing this unless there's further comment

closed 2020-08-12 by sneakers-the-rat