CohenLabPrinceton/pvp #119: TestReport: Oxygen port connectors

CohenLabPrinceton, 2020-06-16:
- [ ] a) An oxygen inlet connector of the VENTILATORY SUPPORT EQUIPMENT, which is OPERATOR‐detachable without the use of a TOOL, shall comply with ISO 80369‐1:2010.

--

- [ ] b) VENTILATORY SUPPORT EQUIPMENT with this inlet connector shall maintain BASIC SAFETY and ESSENTIAL PERFORMANCE with oxygen supply systems up to 600 kPa, in NORMAL CONDITION.

CohenLabPrinceton, 2020-06-16:
The full ISO 80601 (we don't have the other) specifies a 'RESP-6000' connector. We need a safe way of ensuring the O2 couples to the device and can handle 600 kPa (should be OK), but we need some specs on the actual couplings, hopefully using a part that is certified for 80369 standards.

juliennelachance, 2020-06-16:
We're assuming the existence of a commercial gas blender. So our couplers are ISO compliant: they will be standard O2 DISS fittings (NIST compliant).

CohenLabPrinceton, 2020-06-18:
@juliennelachance specs on our parts that we can list in the table.

closed 2020-10-09 by CohenLabPrinceton