CohenLabPrinceton/pvp #120: TestReport: Certification of standards

CohenLabPrinceton, 2020-06-16:
All VENTILATOR BREATHING SYSTEMS, their parts and ACCESSORIES shall comply with the requirements of this document, whether they are produced by the MANUFACTURER of the VENTILATORY SUPPORT EQUIPMENT or by another entity (“third‐party manufacturer” or healthcare provider).

CohenLabPrinceton, 2020-06-16:
@juliennelachance @sneakers-the-rat I think it might be safe to say something like:

•	Our device is designed to be produced by the healthcare provider or a third-party manufacturer
•	It uses off-the-shelf components or parts that are easy to reproducibly fabricate.
•	It incorporates a built-in testing routine to validate all alarms and basic safety standards.

sneakers-the-rat, 2020-06-30:
not sure what is needed on this one from me

closed 2020-10-09 by CohenLabPrinceton