CohenLabPrinceton/pvp #122: TestReport: Anti-suffocation testing

CohenLabPrinceton, 2020-06-16:
- [ ] c) Under these conditions, the inspiratory and expiratory pressure drop measured at the PATIENT-CONNECTION PORT with all recommended ACCESSORIES in place shall not exceed 6,0 hPa (6,0 cmH2O) at a flowrate of:

- [ ] 1) 30 l/min for VENTILATORY SUPPORT EQUIPMENT intended to provide a DELIVERED VOLUME, Vdel ≥ 300 ml;

- [ ] 2) 15 l/min for VENTILATORY SUPPORT EQUIPMENT intended to provide a DELIVERED VOLUME, Vdel ≤ 300 ml;

CohenLabPrinceton, 2020-06-16:
Need to make sure about the pressure drop when ALL of our HEPA and HMEF etc. are installed. How do we test this? @juliennelachance @zajdel @SophieDvali

juliennelachance, 2020-06-16:
We'll need to measure pressure at the inspiratory and expiratory connections. Maybe Tom can use one of our old 3d printed pressure take-off pieces to take the inspiratory pressure reading. Then the expiratory reading will come from the D-Lite as usual

CohenLabPrinceton, 2020-06-22:
@juliennelachance @zajdel I think this needs to be tested just at the proximal port. We need to measure the pressure drop during inspiration to get to the patient and the pressure drop from the patient to the exp valve essentially.

CohenLabPrinceton, 2020-10-09:
Closing all TestReport issues for cleanup

closed 2020-10-09 by CohenLabPrinceton