CohenLabPrinceton/pvp #123: Optimizing PID control coefficients and performance

mschottdorf, 2020-06-21:
This is to track progress on the dinky, and find the optimal settings to control hardware.

zajdel, 2020-06-21:
Main concern right now is that setting the pressure upstream of the prop valve may not be giving us a high enough flow rate to inflate the lung within 0.2-0.3 seconds, regardless of PIP setting.

CohenLabPrinceton, 2020-06-21:
@mschottdorf @zajdel can you put some waveforms up? Does this depend at all on the inspiratory time setting?

zajdel, 2020-06-24:
Significant progress achieved by using gain scheduling to account for the delay in the pressure propagation. Committed, but not pushed yet. I think I just need to be given push access to manuel's fork.

zajdel, 2020-06-24:
[image]

Showing results from test condition 4 (compliance: 20, resistance: 20, PIP: 35, PEEP: 10)

mschottdorf, 2020-06-24:
I'll implement djc's matlab export, fix J's issues with the PR, then merge your stuff, and we are good :-)

mschottdorf, 2020-10-09:
also done!

closed 2020-10-09 by mschottdorf