CohenLabPrinceton/pvp #133: HAPA correction in controller overwritten

sneakers-the-rat, 2020-08-12:
I'm debugging the erratic_dt controller test and I noticed something weird, I kept getting HAPA warnings, but the set PIP remained at its default 35. When Pressure goes above the hapa_limit (initialized from default values, but updated from alarm_manager in normal usage) for a time longer than cough_duration, the Controller should set valves to close immediately and change the set PIP to 30:

https://github.com/CohenLabPrinceton/pvp/blob/a4d46449d1bf73e465314e4172f2edeb0062c512/pvp/controller/control_module.py#L500

the alarm test method is called inside the PID update method all inside the mainloop method. Later in the mainloop method, however, a check for updated controls happens that causes the `__SET_PIP` to be overridden:

https://github.com/CohenLabPrinceton/pvp/blob/a4d46449d1bf73e465314e4172f2edeb0062c512/pvp/controller/control_module.py#L913

Since the default number of control loops between those updates is 1, and `COPY_SET_PIP` is never changed by the alarm checking routine, the setting never takes effect.

I'm submitting a commit now to fix that part by just also changing `COPY_SET_PIP`, but i'm also concerned about what happens in the next iteration of the mainloop and don't know enough about how it works to confidently debug it: if PIP is set to 30, but there is still a high pressure problem, then we would be causing a high pressure problem on every breath cycle, correct? It seems to me like `__SET_PIP` should be decremented rather than set to a fixed value -- eg. what if `__SET_PIP` was lower than 30 but there was some other malfunction in the machine? And given the new design of the controller with the gain parameter, shouldn't the gain parameter be decremented as well?

sneakers-the-rat, 2020-08-12:
did the `COPY_SET_PIP` write in this commit https://github.com/CohenLabPrinceton/pvp/commit/57dc87e13b34b5e38f9289ff3f5e51f78e8b1374

mschottdorf, 2020-08-13:
Hm, the logic of the old thing was: if HAPA, open the exp. valve and close the insp. valve as soon as possible to reduce the excess pressure, and continue with hardcoded PIP of 30. In principle, PIP to 30 could be removed. 

The only really important thing about HAPA is to close inspiration & open expiration to reach a safe pressure. Once that is done, usual cycling should (even has to?) commence. I felt that resetting _all_ parameters to defaults might be a bit too much, so I touched only the PIP setting. Gain could also be reduced, I guess.

mschottdorf, 2020-09-05:
has beed removed; and HAPA algorithm simplified

closed 2020-09-05 by mschottdorf