CohenLabPrinceton/pvp #134: erratic_dt_test is correctly failing

sneakers-the-rat, 2020-08-12:
test_erratic_dt is failing now, with very high PIPs being the main issue. The test didn't fail before https://github.com/CohenLabPrinceton/pvp/commit/74f5b7ec28eda70b03965867193eb963fa3407b2

wherein sleeping for `_LOOP_UPDATE_TIME` was reinstated. 

`test_erratic_dt` sets `_LOOP_UPDATE_TIME` to test erratic dts, which wasn't being slept on in the visible past. I confirmed that the problem was the high DTs by changing the max delay in the test to 10ms, which made the test run correctly.

(furthermore i set up this magic way to make controller tests happen instantly by using `ControlModuleSimulator.simulator_dt` that ya didn't take advantage of, but that's beside the point :p)

sneakers-the-rat, 2020-08-12:
updated erratic_dt test to use simulated dt and wait for a specific number of breath cycles rather than a specific amount of time (which, since it was being sampled uniformly in time, had the chance of unevenly weighting breath cycles if they take different amounts of time, and always underweighted the last breath cycle)
https://github.com/CohenLabPrinceton/pvp/commit/57dc87e13b34b5e38f9289ff3f5e51f78e8b1374

mschottdorf, 2020-09-25:
solved in `f874a6a50820f84bead3a966bb4e36c325dc4d6b`

closed 2020-09-25 by mschottdorf